STM32-base/STM32-base #1: Similar goal, different approach

MauroMombelli, 2018-07-21:
Hello, today i just posted a similar project and a friend pointed out to your project.
The main difference I see is the usage of a C-based startup script, that make clearer the setup script, and also the vector table is just a C array with a bit of compiler magic to be placed in the right section. Again i believe that make it much easier to use and understand.

Also i see in your startup script the __init_array section initialization is missing, thus does not support C++.

Finally the make approach is completely different, and i agree your is more "canonical" and suggested.

Let me know if you are interested to merge the C startup based approach.

ThomasGravekamp, 2018-07-21:
Hi Mauro,

Thank you for opening this issue. Although I do agree that the C startup code is easier to understand, I'm happy with the startup code the way it is now. If someone wants to know how the startup code works, they can refer to the README.md in the startup code folder, which guides them through the code step-by-step. Besides that, I do not expect to make a lot of (large) changes to the startup code from now on. Anything else than memory initialization can be handled in either the `SystemInit` function or at the start of the `main()` function.

You're right about the `__preinit_array` and `__init_array` C++ initialization sections missing from the startup code. I will add them as soon as I can. I'm currently also missing the `PROVIDE ( end = . );` and `PROVIDE ( _end = . );` statements in the linker file, causing some std C functions not to work as they depend on the `end` and `_end` symbols set there. Let's just say that there is still some work to be done here and there.

Are you planning on using this project as base for any of your projects?

Cheers, Thomas

MauroMombelli, 2018-07-22:
>Besides that, I do not expect to make a lot of (large) changes to the startup code from now on.

I see, you provide weak function where user can "override" its code, this is nice since I guess the name are standardized and maybe even used by some SPL.

>Are you planning on using this project as base for any of your projects?

Im thinking of incorporating your in my main makefile, but as I am already coding stuff with that structure i guess I may do another time. But gonna put a link to this project as much more complete for board support

ThomasGravekamp, 2018-07-22:
> Im thinking of incorporating your in my main makefile, but as I am already coding stuff with that structure i guess I may do another time. But gonna put a link to this project as much more complete for board support

Cool, thanks for the support. This project could benefit from having some users. I will fix the issues mentioned above in the coming days.

ThomasGravekamp, 2018-08-05:
Small status update:
I've done some reading regarding the `__preinit_array` and `__init_array` arrays. The current startup code calls the `__libc_init_array` function which does exactly the same as your `static_init` function. This means the current setup should support C++.

However, if you compile with the `-nostdlib` option set, the compiler will not include anything related to the standard C/C++ libs. When this is the case, you indeed need to call the functions in the `__preinit_array` and `__init_array` arrays yourself.

I'm still considering which way I want to go with this. I cannot find any good resource on whether it's better to _include_ or _not include_ the standard C/C++ libraries on a microcontroller platform. What I do know is that you must implement `memcmp`, `memset`, `memcpy` and `memmove` when not using standard libraries as the compiler may generate calls to these functions. See the [`-nostdlib`](
https://gcc.gnu.org/onlinedocs/gcc/Link-Options.html) option for more information.

MauroMombelli, 2018-08-05:
This is quite a debate. The main idea is we don't want to use dynamic memory (or at least i don-t) and that is the main reason i also disable the exception.
Now we have the stdlib which have a lot of great stuff that does not require exception or *alloc BUT.
But there is no way to tell. Wich mean getting rid of stdlib is for the best; if you want to use something, you need to check if you can use it anyway, so at that point you can just copy the class you need.

ThomasGravekamp, 2018-08-06:
I see your point. I've actually stumbled upon some really nice explanation on why not to use the standard library [here](https://arobenko.gitbooks.io/bare_metal_cpp/content/compiler_output/nostdlib.html). Based on the arguments given on that page and your arguments, I will continue without the standard libraries.

I also added the following compiler options for C++:

- `-fno-exceptions`
- `-fno-unwind-tables`
- `-fno-rtti`

More information on the disabling exception handling [here](https://arobenko.gitbooks.io/bare_metal_cpp/content/compiler_output/exceptions.html) and on disabling RTTI [here](https://arobenko.gitbooks.io/bare_metal_cpp/content/compiler_output/rtti.html).

Thanks for your input!

closed 2018-08-06 by ThomasGravekamp

MauroMombelli, 2018-08-06:
Hi thomas, i agree with the decision for exception, but not for RTTI ; it
add only literlayy byte to the end compilation, but make possible to use
things like static and dinamic cast of c++. Also if you don't use classes,
you will not pay this penalty.