STM32-base/STM32-base #4: Add support for flashing over serial connection

ThomasGravekamp, 2019-07-13:


gszy, 2019-12-27:
The [STM32F030 Demo Board][1] apparently cannot be flashed with `st-flash` when connected via serial alone, but it’s possible to use the [stm32flash][2] tool. I think it would make sense to support both tools in STM32-base.

[1]: https://stm32-base.org/boards/STM32F030F4P6-STM32F030-DEMO-BOARD-V1.1
[2]: https://sourceforge.net/projects/stm32flash/

DarinelSanguino, 2022-08-25:
@ThomasGravekamp I'm a beginner and I didn't understand very well what you meant with your comment. I see that there is already a 'flash' rule in common.mk and and there is also a flashing guide in https://stm32-base.org/guides/flashing and I don't know what else can be added to that.
Did you mean add support for other flashing tools like OpenOCD and stflash?
Is it that the enhancements were already made but the issue is still open?

gszy, 2022-08-26:
Currently flashing is done using [SWD](https://en.wikipedia.org/wiki/Serial_Wire_Debug), by calling [st-flash](https://github.com/stlink-org/stlink) (see [common.mk:155](https://github.com/STM32-base/STM32-base/blob/d260b616689692bac0411812b37be065b34f3354/make/common.mk#L155-L157)). This issue is about support for flashing using the STM32 embedded boot loader, so a plain serial (UART) link. You can learn more about it from [AN4325 “Getting started with STM32F030xx and STM32F070xx series hardware development”](https://www.st.com/resource/en/application_note/an4325-getting-started-with-stm32f030xx-and-stm32f070xx-series-hardware-development-stmicroelectronics.pdf) and similar documents, as well as from the [stm32flash](https://sourceforge.net/projects/stm32flash/) project.