STM32-base/STM32-base #5: Add an installation script for Unix platforms

ThomasGravekamp, 2019-07-13:
Add an installation script (.sh) for Unix platforms to set up a complete STM32-base workspace.

The script should do the following:

- [ ] Check for the presence of make
- [ ] Check for the presence of git
- [ ] Set up the directory structure
- [ ] Clone repositories

phenom128, 2022-03-25:
Actually my issue is with startup_common.s not be seen.
copy the directory it lists with ls and i can edit it, i 'don't'
what i'd love is the output of 'ls -lR ~/STM32-base' if possible?
not sure if acceptable, but it's bw494540 AT gmail . com

my attempt at a scriopt is below.  my main mod is the do the 7z extraction of the ARM toolchain leaving the tar file in ~/Downloads
many thanks 
stephen

/home/stephen/bin/setup-stm32-base.sh
#!/bin/bash
echo "Remove previous install directories and files"
rm -rf STM32-base
rm -rf STM32
rm -rf projects

echo "Creating directories"
mkdir STM32
mkdir -p STM32/STM32-base
mkdir -p STM32/STM32-base/libraries
mkdir -p STM32/STM32-base/projects
mkdir -p STM32/STM32-base/templates
mkdir -p STM32/STM32-base/tools

#sudo apt-get install make

cd ~/Downloads

echo "Using xz archive already extracted"
echo "Unpacking arm toolchain tar archive"
#7z x gcc-arm-11.2-2022.02-x86_64-arm-none-eabi.tar.xz
tar xf gcc-arm-11.2-2022.02-x86_64-arm-none-eabi.tar
echo "Moved toolchain to ~/STM32/STM32-base/tools"
mv gcc-arm-11.2-2022.02-x86_64-arm-none-eabi ~/STM32/STM32-base/tools

# This should result in this directory structure:
cd ~/STM32/STM32-base

# ─┬ STM32
#  └┬  STM32-base
#   ├─ libraries
#   ├─ projects
#   ├─ templates
#   └┬ tools
#    ├┬ arm-none-eabi
#    │└─ ...
#    ├┬ bin
#    │ └─ ...
#    ├┬ lib
#    │└─ ...
#    └┬ share
#     └─ ...

# Install open-source ST-Link Tools
# Install the open-source alternative for ST-Link Tools. Refer to the stlink Github page for installation instructions.
# Step 2: Clone the STM32-base and STM32-base-STM32Cube repositories

# Clone the STM32-base and the STM32-base-STM32Cube repositories in the STM32-base directory.
# my gitpull script is called clone

echo "Clone github repos to ~\/STM32\/STM32 repo\'s to STM32-base & STM32-base-STM32Cube"
cd ~/STM32/STM32-base
clone https://github.com/STM32-base/STM32-base.git
clone https://github.com/STM32-base/STM32-base-STM32Cube.git

# This should result in this directory structure:

# ─┬ STM32
#  ├─ libraries
#  ├─ projects
#  ├┬ STM32-base
#  │└─ ...
#  ├┬ STM32-base-STM32Cube
#  │└─ ...
#  ├─ templates
#  └┬ tools
#   ├─ arm-none-eabi
#   ├─ bin
#   ├─ lib
#   └─ share

# Clone the template repositories
# Clone (one of) the template repositories. 
# A template can be used as starting point for a project based on STM32-base. 
# The following templates are currently available:

# so for me
#     --no STM32-base-F0-template
#    STM32-base-F1-template
#     --no STM32-base-F2-template
#     --no STM32-base-F3-template
#    STM32-base-F4-template
#    STM32-base-F7-template

# Clone (one of) the template repositories inside the templates directory:
echo "Clone (one of) the template repositories STM32F1 inside the templates directory"
echo "Me, F1,F4 \& F7"

cd templates
echo "STM32-base\STM32-base-F1-template"
clone https://github.com/STM32-base/STM32-base-F1-template.git
echo "STM32-base/STM32-base-F4-template"
clone https://github.com/STM32-base/STM32-base-F4-template.git
echo "STM32-base/STM32-base-F7-template"
clone https://github.com/STM32-base/STM32-base-F7-template.git

# This should result in this directory structure:
# ─┬ STM32-base???
#  ├─ libraries
#  ├─ projects
#  ├─ STM32-base
#  ├┬ templates
#  │├┬ STM32-base-F0-template
#  ││└─ ...
#  │├┬ STM32-base-F1-template
#  ││└─ ...
#  │├┬ STM32-base-F4-template
#  ││└─ ...
#  │└┬ STM32-base-F7-template
#  │ └─ ...
#  └┬ tools
#   ├─ arm-none-eabi
#   ├─ bin
#   ├─ lib
#   └─ share

# You can use a template project to test your STM32-base setup. 
# Before you do that, you must create a symbolic link to both the STM32-base
# and STM32-base-STM32Cube directories.
#echo $PWD"<--------------------------------\n"
cd ~/STM32/STM32-base/templates/STM32-base-F1-template
#echo $PWD"<--------------------------------\n"
ln -s ../../STM32-base
ln -s ../../STM32-base-STM32Cube

echo "Finished"

exit

ThomasGravekamp, 2022-03-29:
Hi Stephen,

I fixed the formatting in your first comment. Having a `#` in front of text will make it a header. That's how [markdown](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax) works.

And yes, there are two `STM32-base` directories if you follow the installation guide. One is the one you create anywhere you like, to act as your workspace. The other one is a copy of this repository. Note that the top level `STM32-base` directory does **not** have to have that name.

And another bit of feedback on your installation script: I would like to stay away from using 7zip. I want to limit the number of extra application one has to have installed to be able to run this script.

phenom128, 2022-03-29:
i tend to use midnight commander way too much, convenient but archive extraction is slow.
you might want to check the arm available formats, don't recall seeing tgz or zip under linux(???),else i would have used one;

i extract to a tar file and use that; simply me, way faster and easier whilst working on the script.
what i should do it a copy of an expanded tree.

**the arm compiler extracts into gcc-long-version-string, actually it's the sub-directories need to be copied!**

ok in light of comments, i'll go around it again, mind first i had STM32 at the top, that confused me,, that's on the install page as well.
original issue way not seeing startup_common.s i think, thoughts?

phenom128, 2022-03-29:
succinct version of the lost one.
apologies, i don't do html much less markdown, text, an occasional makefile and then it's usually c or perl.

i'll take another go at it.

Install guide has STM32 as a top level - confused me, also the number of STM32-base ones.
Some clarification over the STM32-base lop level and the other

arm tools, not sure i saw tgz or zip formats listed, i'll check for updates :-)

**arm compiler extracts into directory gcc-rather-long-version-string; but its the sub-directories that need moving?**

The common-mk(?name) makefile, maybe examples of where if something is in the 'wrong place' of how it's changed. I'm thinking about the arm version string directory?

from the output, not seeing startup_common.s, copy the path and ls shows, i can even edit it. but something is not found.
logically suggesting path construction maybe?

couple of decades back I once had a conversation with a volatile gentleman, who thought i was having a real go at him about his code, sorted that! 

so please even if it initially appears to be that way, re-read and accept that i'm never critical, just aspects and suggestions only.
stephen

phenom128, 2022-03-30:
Current last message
stephen@i5:~/my_stm32/templates/STM32-base-F4-template$ make
../../tools/bin/arm-none-eabi-g++ -c -specs=nosys.specs -Wall -Wextra -g -ffunction-sections -fdata-sections -D STM32F407xx -Wa,--defsym,CALL_ARM_SYSTEM_INIT=1 -Wl,-L./STM32-base/linker,-T./STM32-base/linker/STM32F4xx/STM32F407xE.ld -I./inc -I./STM32-base/startup -mcpu=cortex-m4 -march=armv7e-m+fp -mlittle-endian -mthumb -masm-syntax-unified -I./STM32-base-STM32Cube/CMSIS/ARM/inc -I./STM32-base-STM32Cube/CMSIS/STM32F4xx/inc -fno-threadsafe-statics -fno-rtti -fno-exceptions -fno-unwind-tables STM32-base/startup/STM32F4xx/STM32F407xx.s -o obj/startup_STM32F407xx.o
STM32-base/startup/STM32F4xx/STM32F407xx.s: Assembler messages:
STM32-base/startup/STM32F4xx/STM32F407xx.s:1: Error: can't open startup_common.s for reading: No such file or directory
make: *** [STM32-base/make/common.mk:141: obj/startup_STM32F407xx.o] Error 1
stephen@i5:~/my_stm32/templates/STM32-base-F4-template$ 

this is what prompted my first comment
you may notice that its now all under '~/my_stm32'

~/my_stm32/templates/STM32-base-F4-template$ ls -l
total 52
-rw-r--r-- 1 stephen stephen 35147 Mar 30 19:02 LICENSE
-rw-r--r-- 1 stephen stephen   127 Mar 30 19:02 Makefile
drwxr-xr-x 2 stephen stephen  4096 Mar 30 19:48 obj
-rw-r--r-- 1 stephen stephen   399 Mar 30 19:02 README.md
drwxr-xr-x 2 stephen stephen  4096 Mar 30 19:02 src
lrwxrwxrwx 1 stephen stephen    16 Mar 30 19:02 STM32-base -> ../../STM32-base
lrwxrwxrwx 1 stephen stephen    26 Mar 30 19:02 STM32-base-STM32Cube -> ../../STM32-base-STM32Cube

yell if you want the script?
stephen

hopefully it all pasted properly
stephen

bbuyukyuksel, 2025-09-08:
I solved this issue: Actually my issue is with startup_common.s not being seen.
This is the solution:
https://github.com/STM32-base/STM32-base/pull/35