STM32-base/STM32-base #11: Using HAL causes a mass of errors

mbardwell, 2019-10-13:
I tried to use HAL_GPIO_Init() but I get a bunch of warnings like

> STM32-base-STM32Cube/HAL/STM32F4xx/src/stm32f4xx_hal_exti.c: In function 'uint32_t HAL_EXTI_GetPending(EXTI_HandleTypeDef*, uint32_t)':
STM32-base-STM32Cube/HAL/STM32F4xx/src/stm32f4xx_hal_exti.c:374:66: warning: unused parameter 'Edge' [-Wunused-parameter]
 uint32_t HAL_EXTI_GetPending(EXTI_HandleTypeDef *hexti, uint32_t Edge)
                                                                  ^~~~
STM32-base-STM32Cube/HAL/STM32F4xx/src/stm32f4xx_hal_exti.c: In function 'void HAL_EXTI_ClearPending(EXTI_HandleTypeDef*, uint32_t)':
STM32-base-STM32Cube/HAL/STM32F4xx/src/stm32f4xx_hal_exti.c:401:64: warning: unused parameter 'Edge' [-Wunused-parameter]
 void HAL_EXTI_ClearPending(EXTI_HandleTypeDef *hexti, uint32_t Edge)

and then errors like

> /usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: /tmp/ccyMnJ2a.o:/home/mike/Documents/STM32-base/pickleball-f4/STM32-base-STM32Cube/HAL/STM32F4xx/src/stm32f4xx_hal_timebase_rtc_wakeup_template.c:82: multiple definition of `hRTC_Handle'; /tmp/ccUa8rBK.o:/home/mike/Documents/STM32-base/pickleball-f4/STM32-base-STM32Cube/HAL/STM32F4xx/src/stm32f4xx_hal_timebase_rtc_alarm_template.c:81: first defined here
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: /tmp/ccyMnJ2a.o: in function `HAL_InitTick':
/home/mike/Documents/STM32-base/pickleball-f4/STM32-base-STM32Cube/HAL/STM32F4xx/src/stm32f4xx_hal_timebase_rtc_wakeup_template.c:103: multiple definition of `HAL_InitTick'; /tmp/ccUa8rBK.o:/home/mike/Documents/STM32-base/pickleball-f4/STM32-base-STM32Cube/HAL/STM32F4xx/src/stm32f4xx_hal_timebase_rtc_alarm_template.c:96: first defined here

To get the includes working, I modified common.mk by adding

`GCC_FLAGS += -I$(STM32_CUBE_PATH)/HAL/$(SERIES_FOLDER)/inc`
`SRC += $(STM32_CUBE_PATH)/HAL/$(SERIES_FOLDER)/src/*.c`

I also renamed stm32f4xx_hal_conf_template.h to stm32f4xx_hal_conf.h` and my main is as simple as

```
#include "stm32f4xx.h"
#include "stm32f4xx_hal_gpio.h"

void init_motor_driver() {
    GPIO_InitTypeDef gpio_config; // pin, mode, pull, speed, alternate
    gpio_config.Pin = GPIO_PIN_9;
    gpio_config.Mode = GPIO_MODE_AF_PP;
    gpio_config.Pull = GPIO_NOPULL;
    gpio_config.Speed = GPIO_SPEED_FREQ_LOW; // (below 0.4 MHz)
    gpio_config.Alternate = 0x02; // "select AF2 in AFRH"
    HAL_GPIO_Init(GPIOA, &gpio_config);
}

int main (void) {
    init_motor_driver();

    while (1) {
    }

    return 0;
}
```

Any ideas?

ThomasGravekamp, 2019-10-13:
Hi, thanks for reporting this issue.

The warnings are triggered in by code in the HAL. They pop up because of the [-Wall](https://gcc.gnu.org/onlinedocs/gcc-8.3.0/gcc/Warning-Options.html#index-Wall) command set in the `common.mk` file. There currently is no way to disable these warnings other than changing the `common.mk` file.

It looks like you haven't read the [Makefile Options page](https://stm32-base.org/cheatsheets/makefile-options) yet. The only thing you need to do is set the `USE_ST_HAL` variable in your project's makefile. The example on that page tells you to set it to "true" but the actual value does not matter since the makefile only checks whether it is set or not.

The cause of the errors you're encountering lies in the fact that the `src` folders of the HAL contain template files which, as it turns out, shouldn't be included like they currently are.

The way I see it, there are two solutions:

- Move the template files to a separate folder in the STM32-base-STM32Cube package, _or:_
- Filter the "*_template.c" out of the list of files that should be included: `$(shell find $(STM32_CUBE_PATH)/HAL/$(SERIES_FOLDER)/src/*.c ! -name '*_template.c')`

I don't like the first option, since I'm planning on supporting the official STM32Cube repositories. And obviously, I can not change their contents.

I think I'll go ahead with the second option. The only downside is that I'll have to use the make shell function to execute a find command to get a list of paths to every non-template *.c file in the src folder. This may make supporting Windows a bit harder in the future, but this is not a problem that can not be solved. And it will also increase the size of the GCC call since every C source file will be included separately.

I'd like to know what think of this. Maybe you've got a third option that I haven't thought of yet?

mbardwell, 2019-10-17:
I'm not the best with Make. What do you mean "it will also increase the size of the GCC call since every C source file will be included separately"?

Option 2 is reasonable

ThomasGravekamp, 2019-10-19:
>I'm not the best with Make. What do you mean "it will also increase the size of the GCC call since every C source file will be included separately"?

When `make` invokes the `gcc` command, the command is outputted to your terminal. Since every C file will have to be included separately to prevent the templates from being included, it will now put every file and the full path to that file in the compilation flags. This instead of the single wildcard that was there before.

I'll go for the second option. I'll try to push the solution tomorrow.

ThomasGravekamp, 2019-10-20:
It should be fixed now. Thanks for reporting this issue.

commit 7c8c17ee (2019-10-20): Use a find command instead of a wildcard to include the HAL source files. This prevents template files to be included. Fixes #11.
closed 2019-10-20 by ThomasGravekamp