STM32-base/STM32-base #12: Change request: split makefiles into more generic parts

uffejakobsen, 2019-11-13:
Hi,

Change request: split makefiles into more generic parts

First of all thanks for at really nice project :-)

It would be nice if the makefiles was split into more generic parts like:
common-rules.mk
common-config.mk OR common-flags.mk
or similar...

Reason:
I'm using STM32-base but my source files are organized different than yours.
You can say that I'm supplying my own build rules that uses your build flags. 
If your common.mk was split into more generic parts are mentioned above - this would be possible :-)

ThomasGravekamp, 2019-11-13:
Hm, I'm not sure if I fully understand your requirements.

You say that your source files are organized differently. Do you mean your C/C++ source files (.h, .c, .cpp) or other files too?

Is this something that can be solved by adding makefile configuration options, like the ones listed [here](https://stm32-base.org/cheatsheets/makefile-options)? Or are your rules that different?

uffejakobsen, 2019-11-19:
My project have a different directory layout - and consist of a mix og C/C++ source files

Like you I also have a habit of producing compile output out-of-tree - not mixing generated files with sources. But unlike your approach - my build output is a hierarchical based on the subdir locations of the source files...   

What I'm trying to say is that your current makefiles works really great if you layout your project at you have planned :-)
Currently your common.mk just includes everything called src/*.c and right now the only way I can block your makefile from doing so is have SRC_FOLDER name a dummy folder with an empty file named empty.c  :-)

What I'm purposing is that you split your makefile into common-rules.mk and common-flags.mk or similar and the have a backwards-compat convienience wrapper c>

That way you would be able to use as much of your project as possible - while supplying your own makefile rules

I hope is makes more sense :-)

ThomasGravekamp, 2019-11-24:
Sorry for my late answer.

The makefiles still need to be made more configurable. One of the problems is like you described in this issue, it forces a specific project layout with at least a `src` folder containing source files. Another "problem" is that the makefiles also force a certain set of GCC flags to be set.

I like your idea. I agree that the common.mk must stay as a convenience wrapper and as an out-of-the-box default.

I think the makefile should be split up into the following files:

- `common-variables.mk`: Contains variables common to every STM32-base project. Stuff like `STM32_BASE_PATH`. This file should also include the `series-folder-name.mk` and series specific makefiles.
- `common-recipes.mk`: Contains recipes like `clean` and `flash`.
- `default-flags.mk`: Contains GCC flags.
- `default-recipes.mk`: Contains recipes for building a project.
- `common.mk`: Includes all the makefiles listed above. This file can check if a variable like `CUSTOM_RECIPES` is set, so it can skip including the `default-recipes.mk` file.

I currently don't have enough time to make these changes. Feel free to make a PR for these changes. I will be available for further discussion, reviewing and testing.

Also, for bonus points, document which variables are exposed by the `common-variables.mk` file and how to use project level overrides for flags and recipes. Let me know if you feel like doing this.

uffejakobsen, 2019-11-25:
Hi Thomas,
Thanks for your answer.
I'm currently working on modifications to the makefiles - that over time could end up in something like what you have described.
Like you I'm also a little constrained in respect to time.
But when I have something that looks generic enough for public display - I'll send you a snapshot

Maybe we can transform it to something useful over time :-)

ThomasGravekamp, 2019-11-27:
Even if you've only just got an idea on how to implement this, feel free to add it to this thread. Anyhow, thanks for raising this issue. It's hard to predict what people want.