STM32-base/STM32-base #15: bin file is unexpected large size.

AbdUslu91, 2020-04-01:
Hello,
I just add only singe line " #include "stm32f1xx_hal_gpio.h" " in  templates/STM32-base-F1-template/src/main.c file. Enabled hal lib with addin USE_ST_HAL in Makefile. Than hit the make, take some warning but not problem for compiling. But binary file size is very large about 182 kbytes.  How can reduce file size and hide this warning? Thanks

` In file included from ./STM32-base-STM32Cube/HAL/STM32F1xx/inc/stm32f1xx_hal_adc.h:852,
                 from ./STM32-base-STM32Cube/HAL/STM32F1xx/inc/stm32f1xx_hal_conf.h:266,
                 from ./STM32-base-STM32Cube/HAL/STM32F1xx/inc/stm32f1xx_hal.h:46,
                 from STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_adc.c:276:
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_adc.c: In function 'HAL_StatusTypeDef HAL_ADC_Start(ADC_HandleTypeDef*)':
./STM32-base-STM32Cube/HAL/STM32F1xx/inc/stm32f1xx_hal_adc_ex.h:488:5: warning: enumeral and non-enumeral type in conditional expression [-Wextra]
  487 |   (( (((__HANDLE__)->Instance) == ADC1) || (((__HANDLE__)->Instance) == ADC2)  \
      |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  488 |    )?                                                                          \
      |    ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  489 |     (ADC1->CR1 & ADC_CR1_JAUTO)                                                \
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  490 |     :                                                                          \
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  491 |     (RESET)                                                                    \
      |     ~~~~~~~
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_adc.c:811:11: note: in expansion of macro 'ADC_MULTIMODE_AUTO_INJECTED'
  811 |       if (ADC_MULTIMODE_AUTO_INJECTED(hadc))
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_adc.c: In function 'HAL_StatusTypeDef HAL_ADC_Start_IT(ADC_HandleTypeDef*)':
./STM32-base-STM32Cube/HAL/STM32F1xx/inc/stm32f1xx_hal_adc_ex.h:488:5: warning: enumeral and non-enumeral type in conditional expression [-Wextra]
  487 |   (( (((__HANDLE__)->Instance) == ADC1) || (((__HANDLE__)->Instance) == ADC2)  \
      |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  488 |    )?                                                                          \
      |    ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  489 |     (ADC1->CR1 & ADC_CR1_JAUTO)                                                \
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  490 |     :                                                                          \
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  491 |     (RESET)                                                                    \
      |     ~~~~~~~
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_adc.c:1148:11: note: in expansion of macro 'ADC_MULTIMODE_AUTO_INJECTED'
 1148 |       if (ADC_MULTIMODE_AUTO_INJECTED(hadc))
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_adc.c: In function 'HAL_StatusTypeDef HAL_ADC_Start_DMA(ADC_HandleTypeDef*, uint32_t*, uint32_t)':
./STM32-base-STM32Cube/HAL/STM32F1xx/inc/stm32f1xx_hal_adc_ex.h:452:4: warning: enumeral and non-enumeral type in conditional expression [-Wextra]
  451 |  (( (((__HANDLE__)->Instance) == ADC1) || (((__HANDLE__)->Instance) == ADC2)   \
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  452 |   )?                                                                           \
      |   ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  453 |    (ADC1->CR1 & ADC_CR1_DUALMOD)                                               \
      |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  454 |    :                                                                           \
      |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  455 |    (RESET)                                                                     \
      |    ~~~~~~~
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_adc.c:1281:6: note: in expansion of macro 'ADC_MULTIMODE_IS_ENABLE'
 1281 |   if(ADC_MULTIMODE_IS_ENABLE(hadc) == RESET)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
./STM32-base-STM32Cube/HAL/STM32F1xx/inc/stm32f1xx_hal_adc_ex.h:488:5: warning: enumeral and non-enumeral type in conditional expression [-Wextra]
  487 |   (( (((__HANDLE__)->Instance) == ADC1) || (((__HANDLE__)->Instance) == ADC2)  \
      |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  488 |    )?                                                                          \
      |    ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  489 |     (ADC1->CR1 & ADC_CR1_JAUTO)                                                \
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  490 |     :                                                                          \
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  491 |     (RESET)                                                                    \
      |     ~~~~~~~
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_adc.c:1321:11: note: in expansion of macro 'ADC_MULTIMODE_AUTO_INJECTED'
 1321 |       if (ADC_MULTIMODE_AUTO_INJECTED(hadc))
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rcc.c: In function 'void HAL_RCC_MCOConfig(uint32_t, uint32_t, uint32_t)':
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rcc.c:1017:30: warning: missing initializer for member 'GPIO_InitTypeDef::Mode' [-Wmissing-field-initializers]
 1017 |   GPIO_InitTypeDef gpio = {0U};
      |                              ^
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rcc.c:1017:30: warning: missing initializer for member 'GPIO_InitTypeDef::Pull' [-Wmissing-field-initializers]
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rcc.c:1017:30: warning: missing initializer for member 'GPIO_InitTypeDef::Speed' [-Wmissing-field-initializers]
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rtc.c: In function 'HAL_StatusTypeDef HAL_RTC_GetDate(RTC_HandleTypeDef*, RTC_DateTypeDef*, uint32_t)':
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rtc.c:804:30: warning: missing initializer for member 'RTC_TimeTypeDef::Minutes' [-Wmissing-field-initializers]
  804 |   RTC_TimeTypeDef stime = {0U};
      |                              ^
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rtc.c:804:30: warning: missing initializer for member 'RTC_TimeTypeDef::Seconds' [-Wmissing-field-initializers]
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rtc.c: In function 'HAL_StatusTypeDef HAL_RTC_SetAlarm(RTC_HandleTypeDef*, RTC_AlarmTypeDef*, uint32_t)':
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rtc.c:870:30: warning: missing initializer for member 'RTC_TimeTypeDef::Minutes' [-Wmissing-field-initializers]
  870 |   RTC_TimeTypeDef stime = {0U};
      |                              ^
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rtc.c:870:30: warning: missing initializer for member 'RTC_TimeTypeDef::Seconds' [-Wmissing-field-initializers]
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rtc.c: In function 'HAL_StatusTypeDef HAL_RTC_SetAlarm_IT(RTC_HandleTypeDef*, RTC_AlarmTypeDef*, uint32_t)':
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rtc.c:962:30: warning: missing initializer for member 'RTC_TimeTypeDef::Minutes' [-Wmissing-field-initializers]
  962 |   RTC_TimeTypeDef stime = {0U};
      |                              ^
STM32-base-STM32Cube/HAL/STM32F1xx/src/stm32f1xx_hal_rtc.c:962:30: warning: missing initializer for member 'RTC_TimeTypeDef::Seconds' [-Wmissing-field-initializers]
../../tools/bin/arm-none-eabi-objcopy -O binary bin/stm32_executable.elf bin/stm32_bin_image.bin `

ThomasGravekamp, 2020-04-02:
Hi, thanks for creating this issue.

There seem to be two separate problems here. The warnings are thrown because STM32-base uses the highest warning settings available for GCC. These are set in the `common.mk` file in this repository. You may want to set these to a "lower" warning value if you want to get rid of them. More information on the available options: https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#Warning-Options

Note that I do _not_ support the code in the STM32-base-STM32Cube repository. It is just there as an easy way to _only_ get the CMSIS and HAL files. 

---

Those warnings aside, that binary file size is interesting. You can get that file size down with some compiler and linker tricks as it turns out after fiddling around with that. I was able to generate a binary of 1932 bytes in size.

I added these lines to my project's makefile:

```
GCC_FLAGS = -O1
GCC_FLAGS += -ffunction-sections -fdata-sections
GCC_FLAGS += -Wl,--gc-sections
```

These lines will enable optimizations and the linker will clean up unused sections. I've found this tip over here: https://stackoverflow.com/questions/15314581/g-compiler-flag-to-minimize-binary-size

I will consider whether one of these flags need to be set by default. Or whether I will have to add this to the docs somewhere.

AbdUslu91, 2020-04-04:
Hi Thomas,
First of all thanks for answering. -ffunction-sections, -fdata-sections, -Wl those flags already added common.mk file. When add --gc-sections this flag, compiler give "unrecognized command line option" error. Than I searched for this error. Cause of problem is gcc version higer than 5.3. I am using latest version.

Than i found stm32 example project with makefile. When i build with this makefile and lastest gcc compiler, bin size is nomal about 9k.

ThomasGravekamp, 2020-04-04:
This looks like you added `-Wl` and `--gc-sections` as separate flags to the GCC call. Be sure to add the flag like this: `-Wl,--gc-sections`. Note that there is a `,` between the flags.

The binary size you have got right now is because you enabled optimizations, with the `-O0` flag in your example. I got similar results after adding only the optimization flags.

AbdUslu91, 2020-04-05:
Hi Thomas,
I added the line `-Wl,--gc-sections`  as you said problem solved. Actually size of binary decreased than complied with default setting Thanks for helping :)

closed 2020-04-05 by AbdUslu91