STM32-base/STM32-base #16: EXTI0_1_IRQHandler not called

sbsoftware, 2020-05-15:
I am new to this all and playing around with your framework on a STM32F072 Discovery board. I tried to handle the interrupt that gets fired when pushing the user button on the board. My code works (the LEDs blink as defined in `program1`) until I push the button. Then everything comes to a halt - as if the interrupt default handler is executed instead of my EXTI0_1_IRQHandler procedure. I have spent hours searching for answers but actually most people seem to use other frameworks and I am out of ideas now. What am I missing?

Here's my code:

```
#include "stm32f0xx.h"
#include <cstdlib>

// Quick and dirty delay
void delay (unsigned int time) {
  for (unsigned int i = 0; i < time; i++)
    for (volatile unsigned int j = 0; j < 2000; j++);
}

void resetLEDs () {
  // Reset the state of LED pins to output low
  GPIOC->BSRR = GPIO_BSRR_BR_6 | GPIO_BSRR_BR_7 | GPIO_BSRR_BR_8 | GPIO_BSRR_BR_9;
}

void program1 () {
  resetLEDs();
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_6;
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_7;
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_8;
  delay(300);
}

void program2 () {
  resetLEDs();
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_8;
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_7;
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_6;
  delay(300);
}

static int *ptr = (int *) malloc(sizeof(int));

void EXTI0_1_IRQHandler (void) {
  if (EXTI->PR & EXTI_PR_PR0) {
    EXTI->PR |= EXTI_PR_PR0;

    if (*ptr == 1) {
      *ptr = 2;
    } else {
      *ptr = 1;
    }
  }
}

int main (void) {
  // Turn on the GPIOC peripheral
  RCC->AHBENR |= RCC_AHBENR_GPIOCEN;
  // Turn on the GPIOA peripheral
  RCC->AHBENR |= RCC_AHBENR_GPIOAEN;
  
  EXTI->IMR |= EXTI_IMR_IM0;
  EXTI->RTSR |= EXTI_IMR_IM0;

  NVIC_SetPriority(EXTI0_1_IRQn, 2);
  NVIC_EnableIRQ(EXTI0_1_IRQn);

  // Put LED pins in general purpose output mode
  GPIOC->MODER |= GPIO_MODER_MODER6_0 | GPIO_MODER_MODER7_0 | GPIO_MODER_MODER8_0 | GPIO_MODER_MODER9_0;

  *ptr = 1;

  while (1) {
    if (*ptr == 1) {
      program1();
    }
    if (*ptr == 2) {
      program2();
    }
  }

  // Return 0 to satisfy compiler
  return 0;
}
```

ThomasGravekamp, 2020-05-15:
To be honest with you, I'm not entirely sure either. This project mostly consists of the basics that I learned while diving into STM32 microcontrollers. I've never created a full blown application with any of these chips. That being said, there are some things that come to mind.

Are you comfortable with using GDB? You can run your program and pause execution when your program seems to freeze. You should be able to see where this problem comes from, either from the default handler not being overwritten or an error in your own code. This is the first thing I would like you to check.

Another thing that came to mind was marking your `ptr` variable as volatile. I know you need to do this when working on the Arduino platform for variables that you want to use in interrupt handlers.

And by the way, why are you creating a pointer to an integer? What's wrong with simply using a regular integer?

Let me know if you need help getting GDB going.

sbsoftware, 2020-05-16:
I haven't really used GDB yet but I will dive into it. Last time I tried, I didn't manage to make it run my code on the device.

That `ptr` variable is the result of me playing around and not knowing too much about C. I now declared it as a `volatile` regular integer, but nothing changed in the output of my code (the LEDs blink like `program1` tells them until I push the user button, then everything comes to a halt, meaning LEDs that are on stay on but the program does not continue).

I'll come back when I got my GDB running!

ThomasGravekamp, 2020-05-16:
I looked up the commands you need to connect GDB to the debug server that the `st-util` command exposes: https://github.com/stlink-org/stlink/blob/develop/doc/tutorial.md#using-the-gdb-server

That mini guide assumes you're using libopencm3, but it should work fine with STM32-base. I'm not sure about that `load` and `continue` stuff. If I remember correctly you can simply use `break` to pause your program without having to use `load` and `continue`.

sbsoftware, 2020-05-16:
I got my GDB running now. I set a breakpoint in the while loop and was able to set the `ptr` variable via the debugger to change the behavior of the LEDs. I also set a breakpoint at the start of my interrupt handler but this code never seems to be reached. The board still freezes when I push the button.

I realized the keyword `thumb` occurs in the startup file to define the default handler for the interrupts, and it also occurs in common.mk as the `-mthumb` flag for the compiler. I tried to play around with that flag but I don't know what it means and it didn't change anything.

ThomasGravekamp, 2020-05-16:
Alright, so to wrap this up, we can conclude that your `program1` and `program2` functions work as expected. And now that you've made that integer a regular integer, I do not expect that to be broken in any way.

Your device getting stuck can now still mean two things. One is that the default handler is not overwritten. The other one is that you've done something wrong with setting the registers for that interrupt, which cause an error which is not handler and thus ends up in that default handler.

Can you try to place a breakpoint in [startup_common.s file on line 98](https://github.com/STM32-base/STM32-base/blob/master/startup/startup_common.s#L98)? If it does, we'll need to find a way to tell whether the default handler is overwritten or whether you've made a mistake with one of the registers (which you do very quickly, I've spend some days on getting I2C to work with only using registers). 

The `-mthumb` flag forces the compiler to generate thumb code instead of ARM code. Now that I think about it, this flag might be set implicitly by setting the platform to any of the ARM Cortex-M cores, since these only support thumbs. The `.thumb` directive does the same, but on a function level. I left those directives there when I copied this stuff from the libs offered by ST.

sbsoftware, 2020-05-17:
The breakpoint in startup_common.s brought me a bit further. The infinite loop I find myself in belongs to another interrupt default handler:

```
Breakpoint 1, WWDG_IRQHandler () at ./STM32-base/startup/startup_common.s:98
98          b    Infinite_Loop
```

I'm not sure yet what to conclude from this but at least I have a new trace to follow.

ThomasGravekamp, 2020-05-17:
Ah, that helps. One thing we can conclude is that the problem is most likely not caused by STM32-base. Not entirely sure just yet, but I do not expect STM32-base to be the problem.

It's weird that the Window Watchdog interrupt fires when you've not configured it to do so. You may need to dig a bit deeper to find out the real trigger. You might want to look into this StackOverflow thread: https://stackoverflow.com/questions/27623885/stm32-wwdg-interrupt-firing-when-not-configured

I still suspect that there's an error somewhere in writing to some register. You can easily trigger the HardFault exception on your device when writing the wrong stuff to the wrong register.

sbsoftware, 2020-05-17:
PHEW! Okay, so I got it. Actually, my implementation of the ISR never got linked in the vector table. While taking a look at the .elf file I realized there are "two" entries for my interrupt handler:

```
215: 080009f9     2 FUNC    WEAK   DEFAULT    2 EXTI0_1_IRQHandler
[...]
251: 0800073d    80 FUNC    GLOBAL DEFAULT    2 _Z18EXTI0_1_IRQHandlerv
```

And guess what? The one with the correct name is the `DefaultHandler` while the one with the funny prefix/suffix is my own. I had a similar issue when fiddling around with WebAssembly a few years ago so at least I quickly found the solution from there. The thing is, your Makefile uses the `g++` compiler while I was writing pure C code. The g++ linker always mangles with the symbol names because functions can be overloaded in C++. So to make sure my function name is not wrapped in _Z18v characters, I have to embed my code (or at least the interrupt handlers) in an `extern "C" { }` block. And voilá, everything works as expected.

That was a rough one, but I am happy I can go on now with what I wanted to do 3 days ago ;-)

For completeness, here's my working `main.c` now:
```
#ifdef __cplusplus
extern "C" {
#endif

#include "stm32f0xx.h"

// Quick and dirty delay
void delay (unsigned int time) {
  for (unsigned int i = 0; i < time; i++)
    for (volatile unsigned int j = 0; j < 2000; j++);
}

void resetLEDs () {
  // Reset the state of LED pins to output low
  GPIOC->BSRR = GPIO_BSRR_BR_6 | GPIO_BSRR_BR_7 | GPIO_BSRR_BR_8 | GPIO_BSRR_BR_9;
}

void program1 () {
  resetLEDs();
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_6;
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_7;
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_8;
  delay(300);
}

void program2 () {
  resetLEDs();
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_8;
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_7;
  delay(300);
  GPIOC->BSRR = GPIO_BSRR_BS_6;
  delay(300);
}

volatile static int mode;

void EXTI0_1_IRQHandler (void) {
  if (EXTI->PR & EXTI_PR_PR0) {
    EXTI->PR |= EXTI_PR_PR0;

    if (mode == 1) {
      mode = 2;
    } else {
      mode = 1;
    }
  }
}

int main (void) {
  // Turn on the GPIOC peripheral
  RCC->AHBENR |= RCC_AHBENR_GPIOCEN;
  // Turn on the GPIOA peripheral
  RCC->AHBENR |= RCC_AHBENR_GPIOAEN;

  EXTI->IMR |= EXTI_IMR_IM0;
  EXTI->RTSR |= EXTI_IMR_IM0;

  NVIC_SetPriority(EXTI0_1_IRQn, 2);
  NVIC_EnableIRQ(EXTI0_1_IRQn);

  // Put LED pins in general purpose output mode
  GPIOC->MODER |= GPIO_MODER_MODER6_0 | GPIO_MODER_MODER7_0 | GPIO_MODER_MODER8_0 | GPIO_MODER_MODER9_1;

  mode = 1;

  while (1) {
    if (mode == 1) {
      program1();
    }
    if (mode == 2) {
      program2();
    }
  }

  // Return 0 to satisfy compiler
  return 0;
}

#ifdef __cplusplus
}
#endif
```

closed 2020-05-17 by sbsoftware