STM32-base/STM32-base #17: Support for Nucleo STM32F446RE Board

MikeSalzburg, 2020-06-15:
Hi there,

thanks for the project so far!

I'm fiddling with the **Nucleo STM32446RE** board and ran into problems since the linker and startup scripts as well as the make file and demo code is not working out of the box. 

I would like to create a pull request for my changes but I guess that I need to become a contributor first. Note that my changes basically copy the **STM32446xE** stuff but it seems that everything works and matches quite nicely. When reviewing the pull request everything should be clear : - )

Another thing is that I would like to use the HAL layer, e.g. the HAL_Delay(100); method. 
Just setting USE_ST_HAL = true in the make file results in an error:
**In file included from ./STM32-base-STM32Cube/CMSIS/STM32F4xx/inc/stm32f4xx.h:250
fatal error: stm32f4xx_hal_conf.h: No such file or directory**
I was not able to find any information about the HAL support in the STM32-base project. I'm happy to either try to fix this with your information as part of my changes or create a new issue.

Regards,
Andi

ThomasGravekamp, 2020-06-15:
Hi Andi, thanks for reporting this issue.

I would like to know what exactly did not work. I assume you have used the [guides](https://stm32-base.org/guides/) to set up the toolchain to try and get at least a blinking LED working.

You need to fork this repository and make changes on that repository, on a separate branch. You can then create a pull request. See this [tutorial from Github](https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork) on how to do that.

To make the HAL work, you'll need to copy the [stm32f4xx_hal_conf_template.h](https://github.com/STM32-base/STM32-base-STM32Cube/blob/master/HAL/STM32F4xx/inc/stm32f4xx_hal_conf_template.h) file to your project and rename it to stm32f4xx_hal_conf.h. I thought I had documented this, but apparently not...

MikeSalzburg, 2020-06-15:
Regarding the Nucleo board: this led to a compilation error since non of the two 446 versions ended in a valid define inside STs header file.

I already have the changes in a branch but I think that you need to add me as collaborator to make any changes to this repo.

A docu regarding the HAL would be nice. Hopefully, I did not miss it somehow!

ThomasGravekamp, 2020-06-15:
Ah, I see.

There is no way to make someone a contributor. I can only give people write access, but you do not need that to make a pull request. Please refer to the Github documentation I linked to.

I will add a remark to the make file documentation for now.

MikeSalzburg, 2020-06-16:
Okay, I created **2 forks and 2 pull requests** that should have no merge conflicts.

Regarding the HAL support: I created an **inc** folder within my project, copied the stm32f4xx_hal_conf.h in it and renamed it to **stm32f4xx_hal.h**. Then I added **USE_ST_HAL = true** to the **Makefile** and made use of **HAL_Delay(100);** in my main.c just for testing. The project compiles successfully.

BUT: I think that the config file needs to be adapted carefully to the board/processor. I quickly compared the file with the one generated by **STM32CubeIDE** (for exactly the board I'm using) and there are a lot of differences. I still gave it a try and my binary did not work (have not tried to debug it somehow). Without the HAL everything works fine of course. So do you have any additional information about the necessary configuration within the config header file?

MikeSalzburg, 2020-06-16:
Well now I tried to use my **stm32f4xx_hal.h** file created by the IDE where I only have enabled the minimum set of features (just GPIO for the LED on pin 5). Unfortunately, it still does not work either. 

Interestingly enough both your file **stm32f4xx_hal_conf_template.h** as well as my own one compile successfully (a few warnings seem to be different though and git diff shows a few different casts and stuff, so maybe that's due to a different version). The resulting binaries also work fine. But as soon as I make use of HAL_Delay(...) instead of the delay() method in your template code nothing works (LED does not blink).

Have you made use of the HAL successfully in the past? Maybe I'm still missing something fundamental here.

ThomasGravekamp, 2020-06-16:
The files in the STM32-base-STM32Cube have not been updated for more than a year. So it is likely that STM32CubeIDE will generate a slightly different configuration file.

To be honest, I do not have any experience with the HAL. I once decided to support it in STM32 to give people some more flexibility. And I only tested the integration by only compiling a simple test program and never ran it on an actual microcontroller. I spent more time on STM32-base itself and on the boards overview on the website.

One of the first things that comes to mind when you say that stuff does not work any more after using the HAL_Delay function is that there might be something wrong with your clock configuration. If I remember correctly, there are a bunch of clock related values you can change in the configuration file. I assume you do not change the clock settings in your program and are running the device on its HSI oscillator.

I guess that the only way to find out more about the cause of this is debugging the program. You should be able to use GDB for that via the onboard ST-Link.

ThomasGravekamp, 2022-10-28:
Closing this issue due to inactivity.

closed 2022-10-28 by ThomasGravekamp