STM32-base/STM32-base #19: RTOS as a additional library

marville777, 2020-06-18:
Hi, 

First - very good project. It is very helpful for me. I want to do next step to add FreeRTOS. I added to directory libraries. What should I add to common.mk to compile this directory. I'm completely newbie in this. I work in Visual Studio Code. I added to json:

                "${workspaceFolder}/libraries/FreeRTOS/Source/portable/GCC/ARM_CM4F",
                "${workspaceFolder}/libraries//FreeRTOS/Source/include"

Additional I added to #Standard values for project folders:

SRC_LIB ?= ./libraries/FreeRTOS/Source/include
INC_LIB ?= ./libraries/FreeRTOS/Source/portable/GCC/ARM_CM4F

and to #Input files

SRC += $(SRC_LIB)/*c
INC += $(SRC_LIB)/*h

But after make all I have something like  "No rules for create object 'libraries/FreeRTOS/Source/include/*c'

Can you give me some hint in this case?

ThomasGravekamp, 2020-06-21:
Hi, thanks for creating this issue.

Adding the FreeRTOS sources to your VS Code configuration json has nothing to do with compiling the actual project. It will only help VS Code find source files you include in your project.

I am not familiar with compiling FreeRTOS from source on STM32 micro's. But the output from make tells you that creates the object called `libraries/FreeRTOS/Source/include/*c`.

Now bear with me, my understanding of `make` is still a bit rusty at the moment, but I'll try to explain what is happening here. Make is based on rules which consist of a target, one or more dependencies and commands.

A rule defines what should happen (the commands) to create a desired output (the target) and which files need to be there (the dependencies) to create that output. So what happens here is that there is a rule that has a dependency on the files in the `libraries/FreeRTOS/Source/include` directory, but there is no rule that tells make how to make these files.

`Make` can not find these files, so it assumes they need to be created first, but had no rule that specifies how that can be done.  Since these are source files, they simply need to be there. This looks like something is wrong with the path to the FreeRTOS path.

Some things that you may want to look into;

- Make runs from the root of your project. Does your project folder contain the libraries folder or a symlink to it?
- Are you aware of the fact that you use `$(SRC_LIB)` in both the SRC and INC lines in your issue description? If these are copied and pasted from your project, make sure to fix those references too.

ThomasGravekamp, 2022-10-28:
Closing this issue due to inactivity.

closed 2022-10-28 by ThomasGravekamp