STM32-base/STM32-base #21: Possible HAL issue (STM32F042K6)

fortysixandtwo, 2021-01-16:
First of all, thanks for this nice project!

I'm running into a weird problem on a STM32F042K6 when using the HAL:
I have a simple [project](https://gitlab.com/devrtz/STM32F042_UART_test) which basically 
- initializes HAL
- does `i++` in an infinite loop

When I flash my program and start it
- My `SysTick_Handler()` is never called into
- When I interrupt my program the program counter is somewhere in the system memory (0x1fffe2d0)
If I however power cycle my device and flash, these problems do not occur.
I've tried two things:
- Copy my main.c file into a CubeMX generated project (does not show the same issues)
- Try the template for the F0 series (there is no `SysTick_Handler`, so of course it will not call into it, but it also does never exit my loop and jump to the system memory

Below you can find a session in gdb which demonstrates the issue.
```
(gdb) load
Loading section .isr_vector, size 0xc0 lma 0x8000000
Loading section .text, size 0x70cc lma 0x80000c0
Loading section .rodata, size 0x50 lma 0x800718c
Loading section .ARM, size 0x8 lma 0x80071dc
Loading section .init_array, size 0x8 lma 0x80071e4
Loading section .fini_array, size 0x4 lma 0x80071ec
Loading section .data, size 0x438 lma 0x80071f0
Start address 0x08007124, load size 30248
Transfer rate: 18 KB/sec, 3781 bytes/write.
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x1fffe2d0 in ?? ()
(gdb) # power cycling
(gdb) load
Loading section .isr_vector, size 0xc0 lma 0x8000000
Loading section .text, size 0x70cc lma 0x80000c0
Loading section .rodata, size 0x50 lma 0x800718c
Loading section .ARM, size 0x8 lma 0x80071dc
Loading section .init_array, size 0x8 lma 0x80071e4
Loading section .fini_array, size 0x4 lma 0x80071ec
Loading section .data, size 0x438 lma 0x80071f0
Start address 0x08007124, load size 30248
Transfer rate: 18 KB/sec, 3781 bytes/write.
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x080005dc in main () at src/main.c:34
34            ++i;
(gdb) p HAL_GetTick()
$10 = 1805
(gdb) load
Loading section .isr_vector, size 0xc0 lma 0x8000000
Loading section .text, size 0x70cc lma 0x80000c0
Loading section .rodata, size 0x50 lma 0x800718c
Loading section .ARM, size 0x8 lma 0x80071dc
Loading section .init_array, size 0x8 lma 0x80071e4
Loading section .fini_array, size 0x4 lma 0x80071ec
Loading section .data, size 0x438 lma 0x80071f0
Start address 0x08007124, load size 30248
Transfer rate: 18 KB/sec, 3781 bytes/write.
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x1fffe2ce in ?? ()
(gdb) p HAL_GetTick()
$11 = 0
```
I'm a bit at a loss as to what exactly is going on and would be glad if someone knowledgeable could point me in the right direction (or any direction at all ;) ).
My working theory is: Something weird happens and the MCU reboots and ends up booting from system memory.
I think it must reboot, because otherwise I can't explain why the program counter would end up in the system memory.
Could an unhandled exception/interrupt cause the MCU to reboot?

Am I missing something important in my code (which f.e. CubeMX takes care of for me)?

fortysixandtwo, 2021-01-16:
After some more tinkering, I can report that the problem is _NOT_ in CMSIS or HAL,
but in the startup assembly or the linker script. Having overwritten them with the version from cubemx I don't run into the issue from above.
I will now `diff` the files and make a patch/pull request once I figure out the differences.

ThomasGravekamp, 2021-03-05:
Hi @fortysixandtwo, have you ever found the root cause of this issue? Could it be that the issue reported in #22 caused the problems you were having?

fortysixandtwo, 2021-03-05:
From reading the description and looking at the fix it could very well be!

I will investigate and come back to you after some testing (when I find the time)

ThomasGravekamp, 2022-10-28:
Closing this issue due to inactivity.

closed 2022-10-28 by ThomasGravekamp