STM32-base/STM32-base #28: SysTick_Handler not called due to C++ name mangline

d850b, 2022-02-27:
Hi, 
i like the idea of your project - doing everything without "hidden wizardry". 
But i ran into a problem that took me some head-scratching :

In a condensed for it is this: 
I added a SysTick_Handler to your template main.c and initialized it like this:

#include "stm32f4xx.h"

int tcount = 0;

void SysTick_Handler(void){
        tcount ++;
}

int main (void) {
    SysTick_Config(SystemCoreClock  );
    ....

After 1 second, the code got stuck in the Default_Handler, i.e. weak Systick_Handler wasn't overridden.
After a lot of tinkering, i found out that this is due to the usage of arm-none-eabi-g++, instead of arm-none-eabi-gcc, 
as in in a makefile project generated by STM32CUBEMX. 

Reason: g++ treats even "*.c" as c++, i.e. name mangling is applied to Systick_Handler, the override doesn't work. This can be verified by looking at a linker map output. (Name mangling is NOT applied to main(), whyever that is.)  

Cure: Declare SysTick_Handler in main.c as extern "C". 

So, i propose you place a hint at a prominent place in the documentation or even in main.c. Or, switch back to gcc. Or maybe there are some options to g++ to make it treat .c not as c++. Any of this will hopefully prevent more hair problems :-) 

(I am  using this board : https://github.com/WeActTC/MiniSTM32F4x1, in Makefile i use DEVICE = STM32F411xE )

Best regards,
Thomas

d850b, 2022-02-27:
Ah. Sorry. I see this is basically a duplicate of Issue #16 . Still - a hint would be nice. Overriding interrupt handlers is what most people try first...

ThomasGravekamp, 2022-03-14:
Hi Thomas, thanks for reporting this issue. It indeed seems to be a duplicate of issue #16.

To properly solve this I'll have to look into using gcc instead of g++. I'm a bit short on time at the moment, but I'll leave this issue here as a reminder.