STM32-base/STM32-base #29: stmf030f4p6 beginner question

prepstarr, 2022-08-23:
I'm so confused how to start , what do I need to purchase to program this device?

Is the st link also a uart?

I'm sorry I'm trying to figure this out but I am really tired of spending my extremely limited budget to buy the wrong thing....

is this good enough provided I have the cables ? usb c to usb 3.0? I'm just trying to understand... to get started, thanks.

ThomasGravekamp, 2022-08-23:
Hi there! I assume you have got one of the Chinese development boards with an STM32F030F4P6 device on it.

> what do I need to purchase to program this device?

You need the following parts:

 - A development board with an STM32F030F4P6 device
 - A ST-link programmer, either the official one or a Chinese clone
 - Wires to connect the ST-link programmer to the debug port on the development board

> Is the st link also a uart?

Generally, no. There are ways to use the built-in ST-link on the official Nucleo development boards as UART, but this is not supported on any other board. You need some additional parts for that:

- A UART to USB bridge board with, for example, a CP2102 chip
- A suitable USB cable
- Wires to connect the bridge to the development board

> is this good enough provided I have the cables ? usb c to usb 3.0?

I am not sure what you mean by this. You will only need a USB cable from your PC to the official ST-link and optionally the UART to USB bridge board. And all of this stuff works on USB 2.0, so you only have to make sure to have a cable that fits your programmer or bridge, usually a USB-B (programmer) or USB mini/micro connector, and PC, usually a USB-A or for modern laptops USB-C. USB 3.0 and higher are backwards compatible with USB 2.0, so you do not have to worry about that.

You do not need a USB cable for your development board. Even if there is a USB connector on your development board with an STM32F030F4P6 device, it will most likely only provide power to the board since the device does not support native USB.

Please provide me with the following information if you want more information:

 - Which board do you want to work with or are you planning to buy?
 - Which programmer (ST-link) do you have, if any? 
 - Do you have a UART to USB bridge board?
 - Do you plan on using STM32-base or another platform?

You should be able to get started with this stuff on a budget of about 15$ if you are able to shop on, for example, AliExpress.

prepstarr, 2022-08-24:
As far as Ali Express I had a lot of good experiences buying tea pots and tea but I bought a supposed solid state drive hard disk for a premium price possibly slightly less than a retailer would charge and it ended up being an admitted fake, they refused to refund me and I had a hard time dealing with PayPal to get it refunded though it was done in the end I missed out on lectures from the European test symposium and it caused me a great deal of stress. I don't like the Amazon thing or any perceived shortcuts. However, st.com has an online shop and I hope that they send to Canada. This will be my option.

prepstarr, 2022-08-24:
https://estore.st.com/en/stlink-v3minie-cpn.html

Is it this that I need to Program it? Do they sell the wires also that I can probably buy at access electronics... but if they have it also it wouldn't hurt...

prepstarr, 2022-08-24:
No need for a Chinese clone of the Chinese ordered item itself... I don't need to disrupt the supply chain, I am trying to work in the parameters of the system not against it... I have a taste of my own oem from China , and I am not buying containers I am just looking to get started in development... and when I do my own chip foundry stuff, it will have its own proprietary design and purpose... but thank you, it seems like it is affordable as is, st is offering free shipping this month with no minimum size order...

prepstarr, 2022-08-24:
When I bought the ssd card I didn't think I was doing the undercut thing they offered 2 tb ssd I needed 2 tb backup drives. Costco had 1 tb and they didn't have the other ones anymore due to pandemic. Anyways I got that problem solved with excellency from LeNovo...

prepstarr, 2022-08-24:
https://estore.st.com/en/stlink-v3set-cpn.html

What's the difference with this does v3 have compatibility issues? I want to just get the correct item so I don't have more useful but useless to me electronics that I debate recycling so as not to be a hoarder...

prepstarr, 2022-08-24:
There's videos of Linus Torvalds getting ripped off from Best Buy or Ebay with claims of computer components far different than what was expected..

prepstarr, 2022-08-24:
Which board do you want to work with or are you planning to buy?

I don't have specific plans I am trying to familiarize myself with stm electronics.

Which programmer (ST-link) do you have, if any?
Do you have a UART to USB bridge board?

No do not have anything 

Do you plan on using STM32-base or another platform?

Any platform that works and familiarizes me..

ThomasGravekamp, 2022-08-24:
Well, in that case I would buy one of the Nucleo boards like this one: https://estore.st.com/en/nucleo-f030r8-cpn.html

Nucleo boards come with an integrated ST-link which can be used for other boards too by removing two jumpers. Note that this only seems to apply to Nucleo boards with an ST-link V2.0 built-in. So you can also use the Nucleo board's built-in ST-link to program and debug your STM32F030F4P6 board.

prepstarr, 2022-08-26:
Thank you going to try that out ...

ThomasGravekamp, 2022-10-28:
Closing this issue due to inactivity and since this is not really an issue with STM32-base.

closed 2022-10-28 by ThomasGravekamp