STM32-base/STM32-base #31: GPIO pin toggle rate is 2.5 times underrated

al37919, 2022-11-07:
Tested with STM32F030F4P6 (STM32F030 DEMO BOARD V1.1)
The project is strictly based on the template provided for this chip.
The minimalist example (main.c and Makefile) is attached.
Compilation with no CPPFLAGS += -Og option provides 725 kHz output on PA4. Compilation with this option produces 2 MHz output.
If the clocks are properly initialized to 48 MHz the same change results in 4.8 MHz vs 12 MHz difference in meander rate.
If someone is capable to explain WHY???, I'd be quite obliged. However IMHO it is definitely a bug.
Cheers,
Alex.

[bug_report.tar.gz](https://github.com/STM32-base/STM32-base/files/9956177/bug_report.tar.gz)

ThomasGravekamp, 2022-11-08:
I am not sure how much you know about all of this, so I am going to assume you are a beginner to all of this.

This does not look like a bug to me. You are comparing a non optimized binary to a partially optimized binary. It is only logical that the more optimized binary has a higher performance, and thus is able to switch the pin state faster.

You can see it for yourself when you decompile the generated `.elf` file back to assembly code with the command `../../tools/bin/arm-none-eabi-objdump -d bin/stm32_executable.elf > dump.asm`.

This is the unoptimized main function:
```
080005c4 <main>:
 80005c4:	b580      	push	{r7, lr}
 80005c6:	af00      	add	r7, sp, #0
 80005c8:	4b50      	ldr	r3, [pc, #320]	; (800070c <main+0x148>)
 80005ca:	695a      	ldr	r2, [r3, #20]
 80005cc:	4b4f      	ldr	r3, [pc, #316]	; (800070c <main+0x148>)
 80005ce:	2180      	movs	r1, #128	; 0x80
 80005d0:	0289      	lsls	r1, r1, #10
 80005d2:	430a      	orrs	r2, r1
 80005d4:	615a      	str	r2, [r3, #20]
 80005d6:	2390      	movs	r3, #144	; 0x90
 80005d8:	05db      	lsls	r3, r3, #23
 80005da:	681a      	ldr	r2, [r3, #0]
 80005dc:	2390      	movs	r3, #144	; 0x90
 80005de:	05db      	lsls	r3, r3, #23
 80005e0:	2180      	movs	r1, #128	; 0x80
 80005e2:	0049      	lsls	r1, r1, #1
 80005e4:	430a      	orrs	r2, r1
 80005e6:	601a      	str	r2, [r3, #0]
 80005e8:	2390      	movs	r3, #144	; 0x90
 80005ea:	05db      	lsls	r3, r3, #23
 80005ec:	2280      	movs	r2, #128	; 0x80
 80005ee:	0352      	lsls	r2, r2, #13
 80005f0:	619a      	str	r2, [r3, #24]
 80005f2:	2390      	movs	r3, #144	; 0x90
 80005f4:	05db      	lsls	r3, r3, #23
 80005f6:	2210      	movs	r2, #16
 80005f8:	619a      	str	r2, [r3, #24]
 80005fa:	2390      	movs	r3, #144	; 0x90
 80005fc:	05db      	lsls	r3, r3, #23
 80005fe:	2280      	movs	r2, #128	; 0x80
 8000600:	0352      	lsls	r2, r2, #13
 8000602:	619a      	str	r2, [r3, #24]
 8000604:	2390      	movs	r3, #144	; 0x90
 8000606:	05db      	lsls	r3, r3, #23
 8000608:	2210      	movs	r2, #16
 800060a:	619a      	str	r2, [r3, #24]
 800060c:	2390      	movs	r3, #144	; 0x90
 800060e:	05db      	lsls	r3, r3, #23
 8000610:	2280      	movs	r2, #128	; 0x80
 8000612:	0352      	lsls	r2, r2, #13
 8000614:	619a      	str	r2, [r3, #24]
 8000616:	2390      	movs	r3, #144	; 0x90
 8000618:	05db      	lsls	r3, r3, #23
 800061a:	2210      	movs	r2, #16
 800061c:	619a      	str	r2, [r3, #24]
 800061e:	2390      	movs	r3, #144	; 0x90
 8000620:	05db      	lsls	r3, r3, #23
 8000622:	2280      	movs	r2, #128	; 0x80
 8000624:	0352      	lsls	r2, r2, #13
 8000626:	619a      	str	r2, [r3, #24]
 8000628:	2390      	movs	r3, #144	; 0x90
 800062a:	05db      	lsls	r3, r3, #23
 800062c:	2210      	movs	r2, #16
 800062e:	619a      	str	r2, [r3, #24]
 8000630:	2390      	movs	r3, #144	; 0x90
 8000632:	05db      	lsls	r3, r3, #23
 8000634:	2280      	movs	r2, #128	; 0x80
 8000636:	0352      	lsls	r2, r2, #13
 8000638:	619a      	str	r2, [r3, #24]
 800063a:	2390      	movs	r3, #144	; 0x90
 800063c:	05db      	lsls	r3, r3, #23
[... 141 lines cut ...]
 8000602:	619a      	str	r2, [r3, #24]
 8000604:	6199      	str	r1, [r3, #24]
 8000606:	619a      	str	r2, [r3, #24]
 8000608:	6199      	str	r1, [r3, #24]
 800060a:	619a      	str	r2, [r3, #24]
 800060c:	6199      	str	r1, [r3, #24]
 800060e:	619a      	str	r2, [r3, #24]
 8000610:	6199      	str	r1, [r3, #24]
 8000612:	619a      	str	r2, [r3, #24]
 8000614:	6199      	str	r1, [r3, #24]
 8000616:	619a      	str	r2, [r3, #24]
 8000618:	6199      	str	r1, [r3, #24]
 800061a:	619a      	str	r2, [r3, #24]
 800061c:	6199      	str	r1, [r3, #24]
 800061e:	619a      	str	r2, [r3, #24]
 8000620:	6199      	str	r1, [r3, #24]
 8000622:	619a      	str	r2, [r3, #24]
 8000624:	6199      	str	r1, [r3, #24]
 8000626:	619a      	str	r2, [r3, #24]
 8000628:	e7d9      	b.n	80005de <main+0x1a>
 800062a:	46c0      	nop			; (mov r8, r8)
 800062c:	40021000 	.word	0x40021000
```

The size of both main functions should already reveal which one is faster. Upon quick inspection it looks like the unoptimized version creates the value to put in the `GPIOA->BSRR` register, while the optimized version does that only once. It can be this simple.

al37919, 2022-11-08:
Dear Thomas,
thanks a lot for the exhaustive explanation of the question "why?". You're, of course, right I'm entering the world of STM32, however with some previous experience (avr-gcc + avr-libc + datasheet). The first impression is overwhelming (CMSIS, SPL, HAL, LL). The clear documentation I've found yet only for CMSIS case (ref.man.) I created two basic projects (init clocks, GPIO meander, USART), the one using STM32-base+CMSIS, another one using CubeMX. Here appeared 2.5-times difference in meander rate. I checked all possibly involved registers, and made them to be equal. After that I adopted Makefile from CubeMX project to the STM32-base+CMSIS one. Finally, I found that the stupid optimization option is the reason of the problem.
Probably, you're right, It's not a bug, but a kind of "feature". On the other hand, the STM32-base project's intention is to  "... provide you with a simple and easy to use base project ...". In my opinion it might be worth to at least consider a possibility to include the optimization option in the default Makefile. 
Kind regards,
Alex.

ThomasGravekamp, 2022-11-18:
Sorry for my late answer.

I actually would like to keep the makefile as minimal as possible. What this project is lacking in that regard is documentation. Some sort of a guide to get you started with actually using STM32-base and STM32 devices in general once you've "installed" STM32-base. I might consider adding this flag to the template's makefile, that's also the place it should go when you are setting it yourself.

However, I honestly don't see how this is not simple and easy to use. The STM32-base project is intended to be simple to set up and use, it is not meant to be simple in a "it does everything for you" kind of way. It is also about helping people understand what goes on under the hood, instead of abstracting all of that away like CubeMX and the Arduino port for STM32 both do.

I would classify this as user error, simply because of a lack of knowledge, which is not a bad thing by the way. And in that sense STM32-base probably made you understand more about the toolchain than CubeMX did 😀

I'll close this issue now as it has been solved.

closed 2022-11-18 by ThomasGravekamp